SensorsIot/Bluetooth-BLE-on-Arduino-IDE #1: Request explaination

Ilzam, 2018-02-02:
Hi,

Could you explain function or meaning of below coding.

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    if (onoff) {
      Serial.println("Notifications turned on");
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
    }
    else     {
      Serial.println("Notifications turned off");
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOff, 2, true);
    }

    onoff = onoff ? 0 : 1;
  }
`

frownbreaker, 2018-03-17:
This is the ternary operator in C. The code sample you posted "`onoff = onoff ? 0 : 1;`" there is an `=` (assignment) rather than a `== `(compare) before explained the code you asked about you should first understand the ternary operator:

This is the typical textbook example..

```
//Looking at the maximum example

int findMaximum(int a, int b){
    //if a > b, it returns a, if not it returns b
    return (a > b) ? a : b;
} 
```

Ok so now we know what the ? is used for. Now in C to compare something you use "==" and to set something to a value you use "="

So the code you asked about "onoff = onoff ? 0 : 1;" means _"if the value of onoff is true set the value of onoff to  0 otherwise set the value of onoff to 1"_

For the avoidance of doubt: In C true is represented by any numeric value not equal to 0 and false is represented by 0.

Reading the code initially one might think the value of onoff is never changed from its initial value, but on inspection, the assignment is done using the ternary operator. I think many folk might have missed this subtle difference. Excellent question :) Hope this helps?

1043717432, 2018-07-05:
How do you request that you compile it?
I have an error using Arduino1.85

In file included from C:\Users\eric8\Documents\Arduino\BLE_Proximity_Sensor\BLE_Proximity_Sensor.ino:23:0:

D:\homeassistant\arduino-1.8.5\arduino-1.8.5\portable\sketchbook\libraries\ESP32_BLE_Arduino-7951347ed68313d75c367e1f2cce763cb56d1eb2\src/BLEDevice.h:10:23: fatal error: sdkconfig.h: No such file or directory

  #include "sdkconfig.h"

                        ^

Compilation terminated.

Exit status 1

closed 2019-02-25 by SensorsIot