SensorsIot/Bluetooth-BLE-on-Arduino-IDE #2: Changing payload to 1 or 0

jonny190, 2018-03-13:
Sorry for the potentially east question but how can i change the output to be 1 for home and 0 for away?

Also how complex would it be to have the MQTT topic be a per address function eg:

/Tracking/ID:01 1/0
/Tracking/ID:02 1/0

frownbreaker, 2018-03-17:
If you look at lines 41 and 42

```
#define TOPIC "BTLE"  // Change for your own topic
#define PAYLOAD "1"    // change for your own payload
```

You can change your code so

define TOPIC "BTLEDEV1"  // for the code on Scanner Device one

define TOPIC "BTLEDEV2"  // for the code on Scanner Device two

You can set define PAYLOAD to be whatever works for you.

If by per address you meant per BTLE device, **_not_** per scanner then the best approach would be to send the detected devices address as the payload and process on the MQTT end point e.g. in node-red

frownbreaker, 2018-03-24:
I've noticed some errors occasionally..

Normally the scan runs and connects to WiFI to deliver the MQTT payload e.g.

`BLE Scan restarted.....
BLE Advertised Device found: Name: , Address:X, manufacturer data: Y, serviceUUID: Z
BLE Advertised Device found: Name: X, Address: Y, manufacturer data: Z
Device found: -92
X
`
Other times there will be an error, reboot and then things work fine. Is there a memory leak?

`
BLE Advertised Device found: Name: , Address:X, manufacturer data: Y, serviceUUID: Z
BLE Advertised Device found: Name: X, Address: Y, manufacturer data: Z
Device found: -87
x
CORRUPT HEAP: multi_heap.c:370 detected at 0x3ffe5768
abort() was called at PC 0x4008a747 on core 1

Backtrace: 0x4008afa0:0x3ffde4a0 0x4008b09f:0x3ffde4c0 0x4008a747:0x3ffde4e0 0x4008ac86:0x3ffde500 0x40083f38:0x3ffde520 0x40083f69:0x3ffde540 0x40087369:0x3ffde560 0x4000beaf:0x3ffde580 0x401762e3:0x3ffde5a0 0x40179941:0x3ffde5c0 0x40179fff:0x3ffde5e0 0x4017a049:0x3ffde610 0x400d4e72:0x3ffde630 0x400d4ffd:0x3ffde660 0x400d50d5:0x3ffde680 0x400d2ec9:0x3ffde6d0 0x4019078a:0x3ffde700

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Starting Arduino BLE Client application...
 - Created client

BLE Scan restarted.....
BLE Advertised Device found: Name: , Address:X, manufacturer data: Y, serviceUUID: Z
 [0;31mE (7106) BT: btc_search_callback  BLE observe complete. Num Resp 2
 [0m
off

BLE Scan restarted.....
 [0;31mE (10110) BT: btc_search_callback  BLE observe complete. Num Resp 2
 [0m
off

BLE Scan restarted.....
BLE Advertised Device found: Name: , Address:X, manufacturer data: Y, serviceUUID: Z
BLE Advertised Device found: Name: X, Address: Y, manufacturer data: Z
Device found: -92
X
`

closed 2019-02-25 by SensorsIot