SensorsIot/Bluetooth-BLE-on-Arduino-IDE #3: Fix for "sketch is too big" on ESP32 boards that are new / poorly supported in ESP32 Arduino IDE

frownbreaker, 2018-03-17:
Hi

I've installed the ESP32 support in my Arduino IDE and can compile and run the blink and BLE_Scan examples from the IDE. Both work fine. When I try and compile this project I get an error saying the sketch is too big? 

Archiving built core (caching) in: C:\Users\[USER]\AppData\Local\Temp\arduino_cache_768145\core\core_espressif_esp32_WeMosBat_FlashFreq_80,UploadSpeed_921600_89e97c72e9bd6fa98e6bdd6fae23ba9b.a
Sketch uses 1395266 bytes (106%) of program storage space. Maximum is 1310720 bytes.

Global variables use 93296 bytes (31%) of dynamic memory, leaving 201616 bytes for local variables. Maximum is 294912 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board "WeMos" WiFi&Bluetooth Battery.

https://www.aliexpress.com/item/WEMOS-ESP32-ESP-32S-Wireless-WiFi-Bluetooth-Development-Board-With-18650-Battery-Holder-ESP8266-ESP-WROOM/32841683088.html

For me the fix was for the board: It seems that the default.csv is wrong for some boards

```
# Name,   Type, SubType, Offset,   Size, Flags
nvs,      data, nvs,     0x9000,   0x5000,
otadata,  data, ota,     0xe000,   0x2000,
app0,     app,  ota_0,   0x10000,  0x340000,
eeprom,   data, 0x99,    0x350000, 0x1000,
spiffs,   data, spiffs,  0x351000, 0xAF000,
```

and boards.txt to have modified values of **`.upload.maximum_size=3407872`**

frownbreaker, 2018-03-17:
Closing as this is not a problem with this project but a platform issue: Lots of issues raised already by others at Expresif ESP32 Arduino https://github.com/espressif/arduino-esp32/issues/1075 and elsewhere https://github.com/1technophile/OpenMQTTGateway/issues/181 even nkolban github e.g. https://github.com/nkolban/esp32-snippets/issues/441 so the devs **are aware** and looking at options ;) its not a simple or quick fix. Quoting nkolban _"I think from an ESP32 perspective, when we start to use ESP32 ... the cost "is what it is" and there is little that this library (which is a wrapper around the base ESP32 BLE libraries) can do."_ from  https://github.com/nkolban/esp32-snippets/issues/441

I hit this problem with some of my ESP32 boards so I thought I'd post what I had to do to fix

Thanks again for great content!

closed 2018-03-17 by frownbreaker

henrikaz, 2020-01-03:
Arduino IDE: Tools -> Partition Scheme:change "Default" to "Huge APP(3MB No OTA)"
Source: https://github.com/espressif/arduino-esp32/issues/1075 (User namanbhardwaj42 comment)

naushadck, 2020-11-30:
I was flashing esp32 using examples->camera webserver. I was getting similar error(sketch is too big).
Just followed the above. And resolved it..........................

   "Arduino IDE: Tools -> Partition Scheme:change "Default" to "Huge APP(3MB No OTA)""

eKselent-Tech, 2021-10-06:
This Solved my Problem too.

Arduino IDE: Tools -> Partition Scheme:change "Default" to "Huge APP(3MB No OTA/1MB SPIFFS)"

nholmes7, 2023-05-17:
I was having the same issue with an ESP32 WROVER and the camera webserver example and this solved it.

"Arduino IDE: Tools -> Partition Scheme:change "Default" to "Huge APP(3MB No OTA)""

LiamIverson, 2023-05-30:
> Arduino IDE: Tools -> Partition Scheme:change "Default" to "Huge APP(3MB No OTA)" Source: [espressif/arduino-esp32#1075](https://github.com/espressif/arduino-esp32/issues/1075) (User namanbhardwaj42 comment)

I love you

tomdesignstudio, 2025-06-10:
I was soo frustrated with this issue trying to upload one of the HomeSpan projects onto ESP 32 (ESP32 DEVKIT V1), I flashed new firmware etc. and then I found your solution. So, I just created account here to say BIG THANK YOU!!!!!!!