SensorsIot/Bluetooth-BLE-on-Arduino-IDE #5: Sketch too big causing error compiling

kevinshane, 2018-03-25:
Hi all, I would like to use a esp32 dev board for this sketch, but it keeps telling me sketch is too large
```
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\xxl\AppData\Local\Temp\arduino_cache_910853\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_c4d977c58ddc9dc3878ead19342d1de7.a
Sketch uses 1395498 bytes (106%) of program storage space. Maximum is 1310720 bytes.

Global variables use 94176 bytes (31%) of dynamic memory, leaving 200736 bytes for local variables. Maximum is 294912 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board ESP32 Dev Module.
```

How did you guys flash this sketch to your esp32? What should I do to reduce the size? (it looks not that big to me)

kevinshane, 2018-03-25:
sorry, dulipcated issue, will close

closed 2018-03-25 by kevinshane

1043717432, 2018-07-05:
In file included from C:\Users\eric8\Documents\Arduino\BLE_Proximity_Sensor\BLE_Proximity_Sensor.ino:23:0:

D:\homeassistant\arduino-1.8.5\arduino-1.8.5\portable\sketchbook\libraries\ESP32_BLE_Arduino-7951347ed68313d75c367e1f2cce763cb56d1eb2\src/BLEDevice.h:10:23: fatal error: sdkconfig.h: No such file or directory

 #include "sdkconfig.h"

                       ^

compilation terminated.

exit status 1