SensorsIot/Bluetooth-BLE-on-Arduino-IDE #6: Missing Wifi connection?

kevinshane, 2018-03-25:
Hi all, I've successfully flashed the code to a esp32 dev board, in terminal it always shows:
```
BLE Scan restarted.....
BLE Advertised Device found: Name: iTAG            , Address: ff:ff:c3:0c:91:99, appearance: 961, serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb, txPower: 0
[0;31mE (73134) BT: btc_search_callback  BLE observe complete. Num Resp 1
[0m
off
```
the same words repeated again and again..

It seems the esp32 didn't connect to wifi at all(and mqtt of cuz).. Have I done something wrong?
many thanks, please help

DevThePhoenix, 2018-07-09:
Same problem here have you found any solution to this !
{
Hi all, I've successfully flashed the code to a esp32 dev board, in terminal it always shows:

BLE Scan restarted.....
BLE Advertised Device found: Name: iTAG            , Address: ff:ff:c3:0c:91:99, appearance: 961, serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb, txPower: 0
�[0;31mE (73134) BT: btc_search_callback  BLE observe complete. Num Resp 1
�[0m
off
the same words repeated again and again..

It seems the esp32 didn't connect to wifi at all(and mqtt of cuz).. Have I done something wrong?
many thanks, please help
}

SensorsIot, 2019-02-25:
Unfortunately I cannot do remote debugging :-(

closed 2019-02-25 by SensorsIot

ralong777, 2021-09-21:
Did You find a fix for this?