SensorsIot/Bluetooth-BLE-on-Arduino-IDE #9: Sketch too big  -  Compiled size too large

zimpower, 2018-07-22:
Thanks for the sketches / videos btw - very helpful.

I tried your sketch (and other sketches). When trying to combine BLE + MQTT on my WEMOS  I get the error below. 

Is this something anyone else has come across and solved?

> Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
> Error compiling for board "WeMos" WiFi&Bluetooth Battery.

ohjay93, 2018-07-28:
I get the same error when trying to compile on my Adafruit HUZZAH ESP32 Feather :-(

bcyr, 2018-08-23:
I got the same problem.  Here is the solution for Arduino : 

Tools --> Partition Scheme:  --> Select No OTA (Large APP)

It works for my M5Stack!

closed 2019-02-25 by SensorsIot