SensorsIot/Bluetooth-BLE-on-Arduino-IDE #16: 2.7µA sleep current with an ItsyBitsy  nRF52840 - can't comment on youtube

Brezensalzer, 2022-06-07:
Sorry for using github for communication, but my comments on Your last youtube video (nRF52 boards) are not visible - don't know why,

I have acheived 2.7µA sleep current with an ItsyBitsy nRF52840 and Arduino/Bluefruit. I had to remove the dotstar led and the 
voltage regulator from the board due to high leak currents. On the software side I was unable to put the nordic softdevice to sleep again once started. So I've swapped the logic: Sleep first, then do the measurement/advertising and a soft reset afterwards. For details please have a look at my github repository: https://github.com/Brezensalzer/ble_sensor_adv

Best regards
Stefan

SensorsIot, 2022-06-27:
Interesting! Does this mean that the chip is no more visible and does not transmit any data on BLE?

Brezensalzer, 2022-08-12:
Yes, during sleep the BLE is completely off. It advertizes / broadcast its data for half a second every minute.
My node is running on a 500 mAh LiFePO4 battery since 14.03.2022 with no signs of draining the battery (still on 3.25V).

SensorsIot, 2025-12-05:
Thanks for sharing your low power implementation!